erelke/ha-esolar #6: Setup failed, retrying: list index out of range

thephantom, 2025-04-27:
Hi,

I removed the old version, installed the one from https://github.com/erelke/ha-esolar. After logging in I can select my plant. 

* Select sites to monitor
* click on send
* message "Configuration created for SAJ eSolar." i don't see  the chonge "we found the following devices". My 2 sunplant are visible in https://eop.saj-electric.com/monitor/device.

When I go to the plug in I get an error message. Setup failed, retrying: list index out of range

[image]

thephantom, 2025-04-27:
after upgrading to 2.0.6 i see 1

[image]

erelke, 2025-04-28:
I think this bug has been fixed. If not, please show or send a log file. (home_assistant.log). If it is no longer relevant, please close it.

closed 2025-05-01 by erelke