erelke/ha-esolar #10: Add support for SEC Module

OlorinSledge, 2025-04-28:
Is it possible to add support for the SAJ SEC Module? I can provide data for it as I have one. Thank you.

erelke, 2025-04-29:
For now, the task is to optimize battery management, stability and reliability. New things may come after that. Is the data sent by the SEC module visible in elekeeper? This integration can only handle the data that any module sends to the SAJ servers. If the data is visible in elekeeper, how does it differ from the existing ones?

erelke, 2025-05-01:
I believe the data you are looking for has already been included in the integration for quite some time. Please check whether the options are enabled when you click the 'Configure' button. If they are, you may have up to 30 entities.

erelke, 2025-05-01:
If the data you are looking for is not included, please check the diagnostics file under the 'runtime_data' key to see if you can find them there. If you do, please let me know which ones they are and send me the downloaded file.

OlorinSledge, 2025-05-02:
Thanks for that. I've reviewed the diagnostics file and there are a number of fields that _might_ be the fields I'm after:

todayBuyEnergy
todaySelfConsumpEnergy
todaySellEnergy

However, they all have zero values in them which does not match the elekeeper app on my phone or website. Should all fields be returning data or do you need to "enable" a field first for it to start returning data? I don't even know if these example fields above are the right fields or not.

erelke, 2025-05-03:
If these values are zero, it’s not certain that they are the ones we’re looking for. The data doesn’t need to be 'enabled,' but I compile it from multiple API requests. It’s possible that for the SEC module, there is an additional API endpoint. Interestingly, my 'todaySellEnergy' has a value, even though I don’t have a SEC module. However, the value is incorrect because it matches the total daily energy production—part of which I have consumed, meaning I haven’t sold all of it.

OlorinSledge, 2025-05-05:
Based on what you're saying, I think there must be an additional API endpoint for the SEC module. There was another Home Assistant addon that was previously successfully retrieving the SEC module values, I'm not sure if reviewing its code might help? https://github.com/djansen1987/SAJeSolar. I understand that possibly the entire website has changed, but perhaps it gives some hints on where to get the SEC module info from?

erelke, 2025-05-05:
If you don't see the necessary data in version 2.0.12, I might ask you to share your solar plant data with me on the Elekeeper site. That way, I can see exactly which API endpoints provide different data—just like in issue #7

cmgestevao, 2025-05-05:
> If you don't see the necessary data in version 2.0.12, I might ask you to share your solar plant data with me on the Elekeeper site. That way, I can see exactly which API endpoints provide different data—just like in issue [#7](https://github.com/erelke/ha-esolar/issues/7)

@erelke 
 Hi, I also use a SEC module in a R5 inverter and I'm in the European Node, if you can check i can invite you to my plant on the Elekeeper site. (please send me you e-mail)
I was also using the addon that @OlorinSledge referenced.

erelke, 2025-05-05:
> [@erelke](https://github.com/erelke) Hi, I also use a SEC module in a R5 inverter and I'm in the European Node, if you can check i can invite you to my plant on the Elekeeper site. (please send me [...]

eross _at_ korona.info.hu

Thank you!

OlorinSledge, 2025-05-05:
@cmgestevao nice to meet another SEC module user!
@erelke I did try to add you to my elekeeper instance but it would not allow me, so I'm glad that @cmgestevao has been to. Hopefully this helps get to the bottom of SEC module support! Thank you.

cmgestevao, 2025-05-05:
- platform: saj_esolar
  username: !secret saj_esolar_user
  password: !secret saj_esolar_pass
  provider_domain: eop.saj-electric.com
  plant_id: 0 # Default is 0. Typically ordered in the same way as they are listed in the eSolar app
  sensors: saj_sec # Optional will only work with SAJ Sec Module
  resources:
    - nowPower
    - runningState
    - todayElectricity
    - monthElectricity
    - yearElectricity
    - totalElectricity # Energy -> Solar production
    - todayGridIncome
    - income
    - lastUploadTime
    - totalPlantTreeNum
    - totalReduceCo2
    - status
    - plantuid
    - currency
    - address
    - isOnline
    - peakPower
    - systemPower # Installed capacity

    - pvElec
    - useElec
    - buyElec
    - sellElec
    - buyRate
    - sellRate
    - selfConsumedRate1
    - selfConsumedRate2
    - selfConsumedEnergy1
    - selfConsumedEnergy2
    - plantTreeNum
    - reduceCo2
    - totalPvEnergy
    - totalLoadEnergy # Energy -> Grid consumption
    - totalBuyEnergy
    - totalSellEnergy # Energy -> Return to grid
    #these entities are deprecated since they return incorrect values
    #- totalGridPower # Power being exported to the grid
    #- totalLoadPower # Solar power being currently self-consumed
    #- totalPvgenPower # Power imported from the grid
    #these new entities replace them
    - gridLoadPower # Power imported from the grid
    - solarLoadPower # Solar power being currently self-consumed
    - homeLoadPower # Total power being consumed by the plant (the home)
    - exportPower # Power being exported to the grid

These where the resources avaiable in the addon referenced earlier, if that helps

erelke, 2025-05-06:
I managed to obtain the SEC data and found the necessary modification as well. The existing API endpoints are good, but in the query, I need to provide the SEC device's serial number as a parameter. As soon as I have time, I will add the new entities/attributes. 

@cmgestevao  thanks again for the invitation/sharing.

cmgestevao, 2025-05-06:
Your welcome @erelke , glad to help in this quest

erelke, 2025-05-06:
@cmgestevao  @OlorinSledge 
I have completed a new version. This one already supports SEC and EMS smart meters as well. Several new entities and attributes have been added. But this is not the end—this is just as much as I could fit into tonight. There will be more to come.

[image]

OlorinSledge, 2025-05-07:
Incredible! Can't wait to see what you add next. Thank you so much.

erelke, 2025-05-07:
In version 2.1.2, the meters appeared as standalone devices. They have only one entity, but all their data is available as attributes. I hope you find all the necessary information. If you still need anything, please open a new issue. I'm glad I could help.

closed 2025-05-07 by erelke