erelke/ha-esolar #7: Multiple Inverter Support

MartinDRZ, 2025-04-28:
Hi,

It would appear that the integration is only currently able to get one inverter from the API. Prior to the API changes I was able to see the data for both of my inverters, as well as a total sum for the generation of them etc.

Is this possible with the new Elekeeper API?

Thanks for the great work so far :)

erelke, 2025-04-28:
Can you send me diagnostics data from your hass device?

[image]

erelke, 2025-05-01:
In version 2.0.9, which has been officially released, I believe I have redacted all sensitive data. Please proceed with sending the new diagnostics file now.

MartinDRZ, 2025-05-02:
[config_entry-saj_esolar_air-01JSYKZ3DR9TVZXE9P815Y23SS (1).json](https://github.com/user-attachments/files/20011887/config_entry-saj_esolar_air-01JSYKZ3DR9TVZXE9P815Y23SS.1.json)

File attached - thanks for making the changes.

Looking at it, it doesn't seem to see my R5 in the list at all. It is there in the portal under the same Plant entity. In the previous incarnation, Plant had a set of properties that summed the inverters within the Plant. The SAJ UI does this, so it seems likely that these values are exposed via the API somehow.

erelke, 2025-05-03:
Unfortunately, the diagnostic data does not show a second device. Perhaps, when multiple devices are present, a different API endpoint needs to be queried. It wouldn’t be difficult to investigate, but I would need access to your solar installation data in the Elekeeper app. If you trust me, you can invite me in the installation settings. My email address is eross _at_ korona.info.hu. Of course, you can revoke the sharing at any time.

[image]

MartinDRZ, 2025-05-04:
I have invited you, hopefully this will benefit the community as a whole as I can't be the only person with multiple inverters...

erelke, 2025-05-04:
> I have invited you, hopefully this will benefit the community as a whole as I can't be the only person with multiple inverters...

Thank you! I’ll take a look. By the way, do you also have a SEC module available?

erelke, 2025-05-07:
It looks like handling multiple inverters is working without any issues. Therefore, I am closing this. If you still need any data, please open a new issue. I'm happy to help.

closed 2025-05-07 by erelke