erelke/ha-esolar #14: Battery entities

dusnake, 2025-05-01:
Hi!! Thanks for the last updates! Did you the json that i send you in the other thread? There are very helpful entities ( battery, power... ) that there are very useful for home assistant

RobbieDemaegdt, 2025-05-02:
2.0.0.9 is already an improvement again but indeed the battery entities are still missing.
Like state of charge had a lot of information which I extracted and used on a dashboard.
If needed I can provide you the diagnostics file.

Thanks for the quick rewrite of the previous integration.

dusnake, 2025-05-02:
I correct the information, seeing the option to download home assistant diagnoses, as it puts in your readme, most of the data is already obtained, but it is not imported into home assistant.

Viewing this image from elekeeper with the info that we all normally use basically:

[image]

- Option 1 is already available in HA via:  inverter_xxxxxxxxxx_power
- Option 2 is not in HA, but in the JSON it is in the path: devices.deviceStatisticsData.gridList.gridPowerwatt
- Option3 hasn't been able to figure out how to retrieve that data from the JSON; nothing matches! I don't know if anyone knows how to extract that data.
- Option 4 is in the JSON: devices.deviceStatisticsData.batEnergyPercent

@erelke , could you please update the entities so that Home Assistant can read options 2 and 4?
@RobbieDemaegdt Would you know how to get option 3?

json:

[saj_esolar_air.json](https://github.com/user-attachments/files/20015496/saj_esolar_air.json)

RobbieDemaegdt, 2025-05-02:
I think the total load of the house so number 3 is called totalloadpowerwatt.
It's hard to provide the full path as I'm checking this on my phone.
Can you double check this value if this is also close with your realtime values.

TecnoTalarn, 2025-05-03:
Hello, with version 2.0.9 I don't see any information related to the battery.
I tried to create a template for a sensor to see if I could see them but it doesn't return any data.
I put this into configuration.yaml file

template:
  - sensor:
      - name: "Battery Direction"
        unique_id: inverter_h1s260XXXXXX01135_energy_total_battery_direction
        state: >
          {{state_attr('sensor.inverter_h1s260XXXXXXX1135_energy_total', 'Battery Direction')}}

If I download the diagnostic, I see the battery and the different values, correctly, but they don't appear in the list of sensors.
extract... 
...
    "devices": [
            {
              "deviceSn": "**REDACTED**",
              "emsSn": "",
              "showInPlantHomePage": true,
              "acCouplingFunction": "",
              "afciInfoList": [],
              "aliases": "H1S260XXXXXXX01135",
              "batteryLevelName": "Battery capacity",
              "bluePassword": "",
              "countryCode": "",
              "deviceBatteryInfoBean": null,
              "devicePc": "**REDACTED**",
              "deviceStatisticsData": {
                "acCoupleData": null,
                "acList": [],
                "backUptotalLoadPowerwatt": 0,
                "batCapacity": 0.0,
                "batCapcity": 0.0,
                **"batCurrent": 5.16,**
                **"batEnergyPercent": 97.0,
                "batPower": 213.0,
                "batStatusName": "Discharge",
                "batType": 2,
                "batVoltage": 49.8,
                "batteryList": [],
                "batteryProtocolSet": 4,**
                "currency": "",
                "dataTime": "03/05/2025 07:25:00",
...

Could it be that it doesn't detect my H1s2 correctly?

erelke, 2025-05-03:
dusnake: 

> I correct the information, seeing the option to download home assistant diagnoses, as it puts in your readme, most of the data is already obtained, but it is not imported into home assistant. [...]

For the second option: The _gridList_ contains as many elements as there are connected phases—so these should be summed up. For example, I have three phases. Therefore, the devices.sysGridPowerwatt value could be used for this instead, as it seems to be the total of these. However, I have seen cases where this value is 0, meaning it is not summed. So, I will handle the summation in the code. The fourth option is already included.

RobbieDemaegdt, 2025-05-03:
Is the battery percentage already available in version 2.0.0.9 if so where can I find it back?

erelke, 2025-05-03:
> Is the battery percentage already available in version 2.0.0.9 if so where can I find it back?

Check v.2.0.10

TecnoTalarn, 2025-05-04:
after downloading version 2.0.10 I see that it doesn't load any battery information, among other things.
Looking at the log, I find the following error...

_File "/config/custom_components/saj_esolar_air/sensor.py", line 2588, in native_value
    value = float(plant["raw"]["batEnergyPercent"])
                  ~~~~~^^^^^^^
KeyError: 'raw'_

the truth is that I don't know much about how the program works, but could it be a typing error?

the system detects 36 sensors for me, and it seems that all of them work correctly, except this one:

thanks, for the hard work.

dusnake, 2025-05-04:
> For the second option: The _gridList_ contains as many elements as there are connected phases—so these should be summed up. For example, I have three phases. Therefore, the devices.sysGridPowerwatt [...]

Yes, in my case devices.sysGridPowerwatt appears  as 0, if you need any help tell me and I try to do it as soon as possible.

Fourth option works great!! Thanks a lot!!

@RobbieDemaegdt 
Third option totalloadpowerwatt, it does not look much like the values shown on the web, what is closest is: devices.loadData.systotalloadwatt.

In summary, @erelke 

Data 1 OK
Data 2 you are reviewing it
Data 3 I think it's devices.loadData.systotalloadwatt
Data 4 OK

Erelke I owe you more than 1 beer

RobbieDemaegdt, 2025-05-04:
> Yes, in my case devices.sysGridPowerwatt appears as 0, if you need any help tell me and I try to do it as soon as possible.
>
> Fourth option works great!! Thanks a lot!! [...]

I don't know which kind of system you have but that path for data3 doesn't even exist for me.

For me the total load of my house is in this path:
runtime_data -> plantlist -> selected plant (I have multiple plants) -> devices -> selected device (only 1 in this list) -> deviceStatisticsData -> totalLoadPowerwatt.

This value is for me the exact same one as on the EOP website.

The loaddata part of your path is empty for me.

[image]

erelke, 2025-05-04:
@dusnake What do you think about _devices[].deviceStatisticsData.totalLoadPowerwatt_? Is it the same as _devices[].deviceStatisticsData.loadData.systotalloadwatt_? Maybe I can check both and use the one that exists and is greater than 0. Additionally, I can add as an attribute for this entity any of the following: gridStatusName, totalLoadPowerwattHidden (or devices[].deviceStatisticsData.loadData.systotalloadwattHidden), devices[].deviceStatisticsData.loadData.outCurr...

RobbieDemaegdt, 2025-05-04:
> > Is the battery percentage already available in version 2.0.0.9 if so where can I find it back?
> 
> Check v.2.0.10

I downloaded this version but there is more broken then before.
I see that you have added like the whole json in 1 attribute was this intended?
if so how can I parse that whole string?

[image]

If I have time next week I might check to help out but it's been 15 years since I last did any python programming and didn't make any HA integration yet.

dusnake, 2025-05-04:
> totalLoadPowerwatt

I think so, but I have doubts about whether that property is the right one for the consumption load of the house. In the dump that you generated with Python, was all the info from the elekeeper api? If you know how to extract all the info from the API, even if it's with postman, I try to get all the useful values related to the new elekeeper portal

erelke, 2025-05-04:
> I downloaded this version but there is more broken then before. I see that you have added like the whole json in 1 attribute was this intended? if so how can I parse that whole string?

In the previous version, I added the original data, but I needed to remove it because the attribute size is limited, and all the data doesn't fit.

dusnake, 2025-05-05:
Hahaha, in my case it is the other way around, it has broken my battery entities :), I think erelke controls this casuistry, depending on the type of installation it looks one way or another.

For updating the matter, in relation to the photo:

[image]

Data 1 OK
Data 2 you are reviewing it (sysGridPowerwatt)
Data 3 I think it's devices.loadData.systotalloadwatt (yes, but I have seen that there are always 120W missing that I do not know where they come from, I continue to follow up to confirm it)
Data 4 OK ( for me, in 2.0.10 works, for @RobbieDemaegdt  works in 2.0.12)
Data 4.1 - And to add, in my case the battery charge in W  is:
    "batPower": -1021.0,  (charge)
    "batPower": 516.0, (discharge)

RobbieDemaegdt, 2025-05-05:
I'm able to get all the information but had to dig in both plants to combine it (1 plant is just my solar inverter and my other plant is my battery inverter).
The only thing I mentioned that is missing for my energy dashboard are the battery charge and discharge energy.
Which I was able to find:
Battery Charge: devices[].deviceStatisticsData.todayBatChgEnergy
Battery Discharge: devices[].deviceStatisticsData.todayBatDisEnergy
Both values are in kWh.

[image]

I have the feeling that in the json response there are duplicate values and probably also depending which device you have it's using different properties in the json structure.
It might be overkill to hardcode the type of device in the code and do the parsing according to the type?

TecnoTalarn, 2025-05-05:
[image]

Working... with some minor mods

thanks

RobbieDemaegdt, 2025-05-06:
I made a pull request for the charge and discharge values for my battery.

TecnoTalarn, 2025-05-06:
Hello, I made a fork of the project in order to help identify the sensors

The original code did not detect my Battery, and I saw that the info it retrieves from SAJ indicates that I have a type 0 inverter, and the code seeing this no longer looks for the battery.

**_if ("hasBattery" in device and device["hasBattery"] == 1) or "hasBattery" not in device:_**

where I changed the 1 to 0.

From here I have been modifying it in order to collect the info from the different sensors that interested me.

I am missing the info on the energy that we send to the network and the one that we collect, and I have not been able to see the items where this information is. If anyone knows, and wants to tell me... thanks.

More info on my fork.
This was made by someone who has only programmed in BASIC, when the commodore64 was the king of the party. Forgive my clumsiness.

dusnake, 2025-05-06:
@TecnoTalarn 

Here is the info:

[image]

Data 1 OK
Data 2 you are reviewing it (sysGridPowerwatt)
Data 3 I think it's devices.loadData.systotalloadwatt (yes, but I have seen that there are always 120W missing that I do not know where they come from, I continue to follow up to confirm it)
Data 4 OK ( for me, in 2.0.10 works, for @RobbieDemaegdt works in 2.0.12)
Data 4.1 - And to add, in my case the battery charge in W is:
"batPower": -1021.0, (charge)
"batPower": 516.0, (discharge)

Si quieres mas info hablamos, que yo soy manchego :)

RobbieDemaegdt, 2025-05-06:
> Hello, I made a fork of the project in order to help identify the sensors
> 
> The original code did not detect my Battery, and I saw that the info it retrieves from SAJ indicates that I have a type 0 inverter, and the code seeing this no longer looks for the battery.
> 
> **_if ("hasBattery" in device and device["hasBattery"] == 1) or "hasBattery" not in device:_**
> 
> where I changed the 1 to 0.
> 
> From here I have been modifying it in order to collect the info from the different sensors that interested me.
> 
> I am missing the info on the energy that we send to the grid and the one that we collect, and I have not been able to see the items where this information is. If anyone knows, and wants to tell me... thanks.
> 
> More info on my fork.
> This was made by someone who has only programmed in BASIC, when the commodore64 was the king of the party. Forgive my clumsiness.

I don't think that is the solution because on my plant the hasbattery is 1 which is correct as 1 stands in most cases for true

TecnoTalarn, 2025-05-06:
> I don't think that is the solution because on my plant the hasbattery is 1 which is correct as 1 stands in most cases for true

https://github.com/TecnoTalarn/ha-esolar-H1-6K-S2/blob/main/config_entry-saj_esolar_air-downloaded.json
if you see my diagnostics file, hasbattery is 0. in my case, I need to change this to activate battery past of software. Maybe has only on my... but.

RobbieDemaegdt, 2025-05-06:
> > > Hello, I made a fork of the project in order to help identify the sensors
> > > The original code did not detect my Battery, and I saw that the info it retrieves from SAJ indicates that I have a type 0 inverter, and the code seeing this no longer looks for the battery.
> > > **_if ("hasBattery" in device and device["hasBattery"] == 1) or "hasBattery" not in device:_**
> > > where I changed the 1 to 0.
> > > From here I have been modifying it in order to collect the info from the different sensors that interested me.
> > > I am missing the info on the energy that we send to the grid and the one that we collect, and I have not been able to see the items where this information is. If anyone knows, and wants to tell me... thanks.
> > > More info on my fork.
> > > This was made by someone who has only programmed in BASIC, when the commodore64 was the king of the party. Forgive my clumsiness.
> > 
> > 
> > I don't think that is the solution because on my plant the hasbattery is 1 which is correct as 1 stands in most cases for true
> 
> https://github.com/TecnoTalarn/ha-esolar-H1-6K-S2/blob/main/config_entry-saj_esolar_air-downloaded.json if you see my diagnostics file, hasbattery is 0. in my case, I need to change this to activate battery part of software. Maybe has only on my... but.

A solution could be that we just check if the object is null or not instead of using that hasbattery property.
I indeed see that you have a battery system but both properties have the value 0.

I find the whole json structure of the API confusing

TecnoTalarn, 2025-05-06:
> [@TecnoTalarn](https://github.com/TecnoTalarn)
>
> Here is the info:
>
> [image]
>
> Data 1 OK Data 2 you are reviewing it (sysGridPowerwatt) Data 3 I think it's devices.loadData.systotalloadwatt (yes, but I [...]

Hola, Gracias! 

sysGridPowerwatt is not a measurement of kWh, it is not energy, it is power. I am looking for the Energy consumed and exported to the electrical network, but I do not see this information in the data dump that is downloaded.
In the version that I had operational before the platform change I had access to this information.
No matter how much I look and look, I cannot see it.
I am looking for the fields where the Import (1.56) and Export (1.07) value of the image appears. 

[image]

RobbieDemaegdt, 2025-05-07:
> > [@TecnoTalarn](https://github.com/TecnoTalarn)
> > Here is the info:
> > [image]
> > Data 1 OK Data 2 you are reviewing it (sysGridPowerwatt) Data 3 I think it's devices.loadData.systotalloadwatt (yes, but I have seen that there are always 120W missing that I do not know where they come from, I continue to follow up to confirm it) Data 4 OK ( for me, in 2.0.10 works, for [@RobbieDemaegdt](https://github.com/RobbieDemaegdt) works in 2.0.12) Data 4.1 - And to add, in my case the battery charge in W is: "batPower": -1021.0, (charge) "batPower": 516.0, (discharge)
> > Si quieres mas info hablamos, que yo soy manchego :)
> 
> Hola, Gracias!
> 
> sysGridPowerwatt is not a measurement of kWh, it is not energy, it is power. I am looking for the Energy consumed and exported to the electrical network, but I do not see this information in the data dump that is downloaded. In the version that I had operational before the platform change I had access to this information. No matter how much I look and look, I cannot see it. I am looking for the fields where the Import (1.56) and Export (1.07) value of the image appears.
> 
> [image]
> 
> or the same values here
> 
> [image]

I also don't find the Import/Export today, no idea if the value is maybe calculated somehow, I didn't really use it in the past as I use my digital electrical meter for those values.

TecnoTalarn, 2025-05-07:
I don't think this information is calculated. If you go to the inverter screen, physically, this information is there. And also, if the remote server had to calculate it, it would have to receive continuous status information every second and that's not the case.
I think that some other source of information is missing, or maybe some of the fields that appear at zero have some other access path.
Since my skills as a programmer are almost nil, I leave this job to the specialists. I've tried to discover how and where the data comes from, but I'm stuck on variables, and my old head can't get there.
The **battery temperature** is also missing as it appears in the app and on the manufacturer's website, so I understand that we're not downloading all the possible fields.
The **export/import today** fields are interesting for the energy control tab, and then being able to compare the data with the distribution company and see what they are swindling us from on each bill.

RobbieDemaegdt, 2025-05-07:
Yeah like I already said, I don't like this json structure, it's not very clear.
Maybe there is another source for that information.

For the battery temperature and see that the value is 0 for me so it's probably also broken/missing.

dusnake, 2025-05-07:
I will update the data after the last great update from @erelke (thanks for the great job you are doing).

-  **In my case the battery percentage has been lost, it is still at:**

devices.batEnergyPercent
"batEnergyPercent": "29",
or

devices.deviceStatisticsData.batEnergyPercent
      "batEnergyPercent": 29.0,

- **Input and output of energy to the battery:**

devices.batteryDirection
"batteryDirection": -1,

or

devices.deviceStatisticsData.batPower
      "batPower": -717.0,

-  I still think the current load is in systotalloadwatt (I don't know if you can include it).

 Tecnotalarn, I think that the properties you indicate do not appear in the json (I don't know if it's because of the api, which doesn't seem to be the best in the world), but that data could be generated from home assitant using the powergrid property, right?

RobbieDemaegdt, 2025-05-07:
> Hi, I have changed this:
> 
> ```
>         if plant["type"] in [1,2,3] and (("hasBattery" in plant and plant["hasBattery"] == 1) or "hasBattery" not in plant):
> ```
> 
> to:
> 
> ```
>          if plant["type"] in [1,2,3] and (("hasBattery" in plant and plant["hasBattery"] == 0) or "hasBattery" not in plant):
> ```
> 
> ["hasBattery"] == 1 to: ["hasBattery"] == 0
> 
> And this way it works for me, looking at the json I have:
> 
> "type: 1 in plantList "hasBattery": 0 in devices
> 
> [saj_esolar_air-01JTN4T6GEF2NQVCB5Y5GYJS2A-H1S2402J2114E00808-873f47f09062b67ce9b1cabe67e56572.json](https://github.com/user-attachments/files/20082202/saj_esolar_air-01JTN4T6GEF2NQVCB5Y5GYJS2A-H1S2402J2114E00808-873f47f09062b67ce9b1cabe67e56572.json)

I understand that this works for you but this would break it for other people so we should fine a solution that works for both.

dusnake, 2025-05-07:
> I understand that this works for you but this would break it for other people so we should fine a solution that works for both.

Of course, its a workaround for my automations in home assistant, if i can help with anything im here

TecnoTalarn, 2025-05-07:
I made this same modification initially (hasbattery) , but I searched for all the places where type selected the type of inverter and added 0, so that it shows me up to 32 sensors, many at 0 or unknown. I know that the idea was to avoid unnecessarily loading HA with sensors that do not contribute anything, but first we need to see which sensors to choose. I, particularly, am not at all interested in knowing the value of "Plant Talarn today Equivalent Hours" for example and it may be useful to others.
I think we should see everything that can be collected, and then do a search for what is really interesting.
In addition, we need to know what happens depending on the firmware version, or the working mode, which surely activates or deactivates data and that is why some of us have some data and others have others, or they are in other places.

dusnake, 2025-05-07:
Works mate!!! Coffee invited!

erelke, 2025-05-07:
> Works mate!!! Coffee invited!

**Glad I could help. Use it for a long time 😜**

closed 2025-05-07 by dusnake