erelke/ha-esolar #15: plant sensors are "0" or unknown

joker5689, 2025-05-02:
[image]

[config_entry-saj_esolar_air-01JT7MM5HRFXSMPX2KWV5DPR06.json](https://github.com/user-attachments/files/20008508/config_entry-saj_esolar_air-01JT7MM5HRFXSMPX2KWV5DPR06.json)

erelke, 2025-05-03:
I don’t know why the Saj isn’t aggregating the data for you. This is how you receive it from the API. The data is available at the device level—enable the option to view device data. Do you have an AS1 battery and no inverter?

joker5689, 2025-05-04:
Hello,

[image]

i only need the batEnergyPercent.

[image]

Thanks a lot for your time so far!

erelke, 2025-05-04:
Yes this data is value of  "Inverter {inverter_sn} Battery State Of Charge" entity. This entity is added if your plant type is **3** (or 1, or 2), and **option "use inverter sensors" is checked**. You can see this entity? Is one from unknow (onbekend?) entities?

[image]

joker5689, 2025-05-04:
"sensor.plant_lamaire_beveren_aan_de_ijzer_state_of_charge" is working again.

Added 1 line of code. I don't now if this the correct way, but it works for me.

[image]

[image]

[image]

erelke, 2025-05-05:
The row you added returns the state of charge of the inverter's battery, not the pant's. The plant's charge level needs to be calculated if all data is available (The plant may contain multiple batteries or a device equipped with a battery). Unfortunately, this is not always the case. The device's battery SOC value (which is also the row you added) is already represented by the ESolarInverterBatterySoC class, which creates the entity named 'Inverter {inverter_sn} Battery State Of Charge.' So, you just duplicated it.

erelke, 2025-05-05:
The main problem is that the structure of data coming from different SAJ solar systems varies. There are also plenty of typos—for example, batCapacity sometimes appears as batCapcity, while in another system it's batCapicity. Madness!

dusnake, 2025-05-05:
> The main problem is that the structure of data coming from different SAJ solar systems varies. There are also plenty of typos—for example, batCapacity sometimes appears as batCapcity, while in another system it's batCapicity. Madness!

yes, version 11 and 12 have broken the type of battery that my plant shows

stews87, 2025-05-06:
[image]

erelke, 2025-05-07:
A lot has changed, and I hope your issue has been resolved as well. If not, please report it in a new post along with diagnostic data. Thank you.

closed 2025-05-07 by erelke

stews87, 2025-05-07:
after update it has divided into 4 devices (2 batteries, system for 'name' and inverter for 'device number') a bit confused. It still remains Load, Sell, Buy at 0kwh. Thanks for your work