erelke/ha-esolar #21: Load,Sell,Buy “0”

stews87, 2025-05-07:
after update it has divided into 4 devices (2 batteries, system for 'name' and inverter for 'device number') a bit confused. It still remains Load, Sell, Buy at 0kwh. Thanks for your work

joker5689, 2025-05-07:
It's the same over here.
4 devices (2 batteries, system for 'name' and inverter for 'device number').

[image]

[image]

[image]

erelke, 2025-05-07:
Thank you for your feedback. Unfortunately, SAJ does not return data in a standardized format for everyone. My data has been displayed correctly from the very first day. Today, many people have had their issues resolved. I'm sorry that it hasn’t worked out for you yet. I will try to find out what might be causing this.
Regarding names: I can only display the data I receive from SAJ. Some devices come with a name, while others have their serial number assigned as a name. Fortunately, in Home Assistant, you can rename any device according to your preferences.

RobbieDemaegdt, 2025-05-09:
I checked the diagnostics data and the plant buy/sell energy is empty for me as well, else I would have solved it.
So no idea if the data is just not returned or it's not in this diagnostics file.

erelke, 2025-05-12:
Please check with the latest version. If the issue still persists, I can help if someone shares their system with me in Elekeeper. I see the "isParallel" flag in the API response, but I don’t know what kind of request is used to fetch the data in this case. I need to verify what happens in such a system. I have an idea of what to expect, but this isn’t a casino. 😃

manducagiuseppe, 2025-05-13:
> Please check with the latest version. If the issue still persists, I can help if someone shares their system with me in Elekeeper. I see the "isParallel" flag in the API response, but I don’t know [...]

How do I share my Elekeeper account with you?

RobbieDemaegdt, 2025-05-13:
which data are you guys missing?

Go into HA and click on download diagnostics file.
You can upload it here and I will have a look which data you don't have and if we can add it.

[image]

manducagiuseppe, 2025-05-13:
> which data are you guys missing?
> 
> Go into HA and click on download diagnostics file. You can upload it here and I will have a look which data you don't have and if we can add it.
> 
> [image]

Thanks for the help, I attach the image of the missing data and the json file of the diagnostics

[image]

[config_entry-saj_esolar_air-01JV2GYM3NYT9761VZGWHYW22F.json](https://github.com/user-attachments/files/20187705/config_entry-saj_esolar_air-01JV2GYM3NYT9761VZGWHYW22F.json)

RobbieDemaegdt, 2025-05-13:
The values that you are missing aren't in that file so that means I can't update the code to have them working.
There might be other API calls that are missing to get this data.
@erelke Any idea if there are other API calls?

manducagiuseppe, 2025-05-14:
> The values that you are missing aren't in that file so that means I can't update the code to have them working. There might be other API calls that are missing to get this data. [@erelke](https://github.com/erelke) Any idea if there are other API calls?

I saw inside the file that there are the entries "todayBatChgEnergy": 0, "todayBatDisChgEnergy": "2.79"  these are imported because I could put them in the energy dashboard, how can I make them display?

RobbieDemaegdt, 2025-05-14:
> I saw inside the file that there are the entries "todayBatChgEnergy": 0, "todayBatDisChgEnergy": "2.79" these are imported because I could put them in the energy dashboard, how can I make them [...]

These are normally already available in the attributes of the battery percentage and you can display they using a helper with template.

[image]

See at the bottom of the screenshot

erelke, 2025-05-14:
> How do I share my Elekeeper account with you?

In your plant settings you have sharing option:

[image]

[image]

erelke, 2025-05-14:
> The values that you are missing aren't in that file so that means I can't update the code to have them working. There might be other API calls that are missing to get this data. [@erelke](https://github.com/erelke) Any idea if there are other API calls?

Yes, I have an idea of which API endpoint we will need in this case. That's why I need to check exactly how it works in this type of system. Unfortunately, there are many different types of plants.

manducagiuseppe, 2025-05-14:
> Yes, I have an idea of which API endpoint we will need in this case. That's why I need to check exactly how it works in this type of system. Unfortunately, there are many different types of plants.

Which email should I share to?

erelke, 2025-05-14:
> Which email should I share to?

eross _at_ korona.info.hu

manducagiuseppe, 2025-05-14:
> eross _at_ korona.info.hu

Done, I'm going to share the plant with you

erelke, 2025-05-14:
Thank you, now I can see your plant as well. As soon as I have time, I will take a closer look, and I hope to determine what needs to be expanded in the integration.

erelke, 2025-05-14:
@manducagiuseppe This was easier than I thought. A new version released. Enjoy ;)

closed 2025-05-14 by erelke