erelke/ha-esolar #24: Q: what is a good update interval

wmwnd, 2025-05-10:
Hi,

The update interval defaults to 10 mins.
Wondering if it possible to check/update every minute, does this hurt the interface api at suppliers side or hit any daily limit? Any thoughts on this, thank you!

douglaswp, 2025-05-12:
I may be wrong, but I believe that the Elekeeper portal receives data from the inverter every 5 minutes.

So there would be no point in reducing the interval between updates, as the data would still be the same.

erelke, 2025-05-12:
@wmwnd  Set the update interval to 1 minute. Then, after half an hour, check the Logbook of the "Plant **** last Upload Time" entity, where you will see how often the elekeeper receives data from your system. It is not worth setting a value lower than this. Most systems typically send data every 10 minutes. However, I have seen update intervals of 1 minute and 5 minutes as well. For certain modules, it may be possible to configure how frequently they send data.

closed 2025-05-12 by erelke