erelke/ha-esolar #35: battery temperature missing

TecnoTalarn, 2025-05-18:
Hello again
I'm looking for the battery temperature parameter.
I've looked and looked at the different values ​​but I've only found the device temperature, but at zero.
If you go to the manufacturer's website, the value does appear when we enter the device's monitoring details. So it must be somewhere.

[image]

How can we find it?
Looking ahead to summer, when the ambient temperature rises, it would be great to be able to monitor the battery temperature remotely, and even be able to create some kind of control to avoid major problems.

Cheers

RobbieDemaegdt, 2025-05-19:
In 1 of the devices on HA there is a diagnostics sensor that displayed the temperature.

[image]

Can you maybe also upload the diagnostics file from HA that you can download for this integration, than erelke or me can have a look at it.

erelke, 2025-05-19:
Do you see your battery as a separate device? If not, please send a file containing diagnostic data. (Simply drag & drop it here in the comment box.)

closed 2025-05-22 by erelke