erelke/ha-esolar #37: AS1-3KS-5.1 batteries, the sell and buy values ​​are still set to 0

Iratxetxo, 2025-05-19:
On my AS1-3KS-5.1 batteries, the sell and buy values ​​are still set to 0. I have two batteries with a total of 10.24kWh. I can see, for example, the temperature of each one, or the % charge.
Should I do something? I don't have the information in the attributes either.
I've looked at and followed #21, but it doesn't work.

[image]

Thanks everyone.

erelke, 2025-05-19:
Can you share a diagnostic file with us? Without it, we're in the dark. Please simply drag & drop it here.

RobbieDemaegdt, 2025-05-20:
I have the same battery and the values aren't in the diagnostics file.
Most of the Buy and Sell values are 0 in the file

erelke, 2025-05-20:
@Iratxetxo Do you only have batteries, or do you also have a solar system? If version 2.1.12 didn’t solve your problem, please share your system with me on the Elekeeper site.

Iratxetxo, 2025-05-21:
@erelke
Hello, updated to version 2.1.12 and it’s not working. I’ve given you access to my plant so you can investigate whatever you need.

In my case, I don’t have an R5 inverter. I manage all the data directly from the battery inverter.
In the diagnostics, I see I have data like:
"nowPower": 1695.0
"pvSelfConsumedRate": 0.743,
"runningState": 1,
"showBatSoc": 1,
"sysGridPowerwatt": 0.0,
"todayBatChgEnergy": 1.41,
"todayBatDisEnergy": 1.02,
"todayPvEnergy": 2.32,
"totalBatChgEnergy": 4683.69,
"totalBatDisEnergy": 3011.23,
"totalFeedinEnergy": 3615.37,
"totalLoadEnergy": 7066.4,
"totalLoadPowerwatt": 2015.0,
"totalLoadSelfConsumedEnergy": 3451.03,
"totalPvEnergy": 6898.21,
"totalPvSelfConsumedEnergy": 5123.49,
"totalSellEnergy": 1774.72,
"unitOfCapacity": "AH",
"updateDate": "21/05/2025 11:05:00",
"yearPvEnergy": 1063.17

It would also be very interesting to try to retrieve the data managed by the battery’s AI, like in the picture:

Or even to be able to send to the API, if possible, to modify the inverter’s status and change its operating mode.

I know it’s a lot, and you’re already doing so much for all of us.
Thank you so much for the amazing work you’re doing! If there’s anything I can help with, I’m here.

erelke, 2025-05-21:
> [@erelke](https://github.com/erelke) Hello, updated to version 2.1.12 and it’s not working. I’ve given you access to my plant so you can investigate whatever you need.

Thank you for sharing. I think I'm on track now, and I'll be releasing a new version soon. 😊

Iratxetxo, 2025-05-22:
Woooooooowwwwwwwww, fantastic work!!!
Now I see tons of data with version 2.2.14.
I'll analyze the data, look at the diagnostics, and if we see that we can include more data, I'll let you know.

Thanks a million!!

closed 2025-05-22 by erelke