erelke/ha-esolar #39: Load home?!?

stews87, 2025-07-03:
congratulations for your work. To be complete I need two entities. home consumption now (Load energyNow) and battery load/discharge now (Charge/Dis charge now). at the moment they are all for total day. thanks

erelke, 2025-07-04:
Thank you for reaching out, I'm happy to help. These values appear in the systems I'm familiar with. In theory, they should appear on your end as well. Please describe what devices you have and send me the diagnostic file that can be downloaded from Home Assistant, so I can see what data we're receiving from the SAJ. There's quite a bit of confusion in the SAJ API—the data from different devices appears differently—so it's possible that you have a combination I haven’t encountered yet.

erelke, 2025-07-04:
These values appear as attributes of the entity named 'Inverter {inverter_sn} Battery State Of Charge'. Please check this entity and its attributes to see whether they are present on your system as well.

erelke, 2025-07-06:
Please send me the diagnostic file:

[image]

stews87, 2025-07-08:
[config_entry-saj_esolar_air-01JTHQ2248NYK9V1JZYERNB0QC.json](https://github.com/user-attachments/files/21115055/config_entry-saj_esolar_air-01JTHQ2248NYK9V1JZYERNB0QC.json)

erelke, 2025-07-27:
I will be able to deal with this request during the autumn.